guymcswain/pigpio-client #3: opened notification socket with handle=0

giminni, 2017-12-21:
I have currently an issue using pigpio-client library inside docker.

My docker container is a privileged container and I am trying to issue a write command to two specific pins over the net. The IP I am using is the docker host IP (172.17.0.1)

`request=  26 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
response=  26 0 0 0 0 0 0 0 0 0 0 0 64 0 0 0
request=  17 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
opened notification socket with handle= 0
response=  17 0 0 0 0 0 0 0 0 0 0 0 130 32 160 0
{
  "host": "172.17.0.1",
  "port": 8888,
  "pipelining": false,
  "commandSocket": true,
  "notificationSocket": true,
  "pigpioVersion": 64,
  "hwVersion": 10494082,
  "hardware_type": 3,
  "userGpioMask": 268435452
}
request=  0 0 0 0 11 0 0 0 1 0 0 0 0 0 0 0
wrote false to pin11
wrote true to pin13
response=  0 0 0 0 11 0 0 0 1 0 0 0 0 0 0 0
deferred request=  0 0 0 0 13 0 0 0 1 0 0 0 0 0 0 0
response=  0 0 0 0 13 0 0 0 1 0 0 0 0 0 0 0
deferred request=  4 0 0 0 11 0 0 0 0 0 0 0 0 0 0 0
response=  4 0 0 0 11 0 0 0 0 0 0 0 0 0 0 0
deferred request=  4 0 0 0 13 0 0 0 1 0 0 0 0 0 0 0
response=  4 0 0 0 13 0 0 0 1 0 0 0 0 0 0 0`

Apparently I cannot get any handle despite the fact I got a valid object with state connected.

giminni, 2017-12-27:
@guymcswain is there any workaround or mitigation?

guymcswain, 2017-12-27:
Sorry that I have just now seen this.  I'll will be looking into this issue
today and will get back to you.

Thanks for using pigpio-client!

guymcswain, 2017-12-27:
Thank you for the debug trace.  I see that a notification handle (0) was
opened.  You performed two gpio.modeSet() commands to pins 11 and 13, then
write 0 and 1 respectively to these pins.  If you read the pin values back
you should observe their levels in the last 32 bits (little endian).  So I
would expect response= 3 0 0 0 13 0 0 0 0 0 0 0 1 0 0 0 after reading pin
13.

Also, the notification handle is used internally to pigpio-client.  It is
used for things like gpio.notify().

Please clarify your question if I have missed your point.

giminni, 2017-12-28:
I connected GPIO pin11 and pin13 to a relay card. If I use node-red-contrib-pigpio I can successful toggle the relay. If I use your library I am unable to open or close the relay contact.

According to the debug list, I see no change of value (request/response shows the same bit array)

My question is the pigpioVersion correct?
Why I cannot toggle the relay writing 0 to pin11 or pin13? 
Is there a workaround for this?

guymcswain, 2017-12-28:
> My question is the pigpioVersion correct?

Yes

> Is handle=0 correct?

Yes.  But this could also be any positive integer.  It's what the OS returns as a file descriptor.

> Do I need pipelining or other flags?

Pipelining settings should not matter to functionality.  It is there more for convenience and possibly performance improvement.

> Why I cannot toggle the relay writing 0 to pin11 or pin13?

Unknown.  What you showed above should work.  Are you connected to pins 23 (GPIO11) and 33 (GPIO13) on the Raspberry Pi 40 pin connector?  Make sure you don't have SPI enabled as GPIO11 has alternate function as SPI_CLK.

guymcswain, 2018-01-04:
@giminni , please let me know if you need additional assistance, have abandoned pigpio-client or otherwise so that I may close this issue.

giminni, 2018-01-04:
@guymcswain 
finally I found my error. 

It was a bit misleading having myPin = GPIO, e.g. pin11 is GPIO(17).
Can you add some notes into your examples, this would reduce hunting bugs where there aren't.

Your library works perfectly, thank for helping me pointing into the right direction.

Keep up the work on this library!

closed 2018-01-04 by giminni