guymcswain/pigpio-client #4: pigpio-client  Do you have automatic reconnection?

bulentkusva, 2018-03-15:
const PigpioClient = require('pigpio-client');
 
 const pi = new PigpioClient.pigpio({host:'192.168.1.231', port:8888,pipelining: false});  

Do you have automatic reconnection?

guymcswain, 2018-03-17:
Please explain the conditions under which automatic reconnection is useful.  I don't think under an error condition it would be useful.

bulentkusva, 2018-03-19:
yes in case of error or if you can not connect to the server you will need to try again

guymcswain, 2018-08-09:
@bulentkusva 
I have created a branch in this repo called 'autoreconnect'.  It will solve 3 cases that I consider important or that I have recently run into:

1. After initial creation of pigpio-client object, if not successfully connected, automatically reconnect if any error occurs.  Likely errors encountered are `ECONNREFUSED` or `EHOSTUNREACH` when the remote pigpiod process or host is not available.  This case is currently covered and has been tested somewhat.

2. After pigpio-client is connected, if a socket receives an error or is forcibly closed, then destroy current connections, emit `'disconnected'` event to application and wait for `pigpio-client.connect()` to reestablish connections.  The code covers this case and I have successfully completed a preliminary test.

3. Detect a dead peer on notification socket using built-in pigpio keep alive packet.  This covers the case when the remote host running pigpiod is unexpectedly reset - say due to a power loss - and comes back up without connections, yet pigpio-client still has open connections (ie no errors have occurred).  The code for this case is not yet written.  I'm currently working on it.

Cases 2 and 3, require a change to your application.  You must monitor for 'disconnected' event and call `connect()` method after any housekeeping.  You'll need to re-initialize pigpio as the state of pins, pwm, alerts, etc have all been reset.  In my app, I just called my `init()` function after attaching a one time listener to the new `'connected'` event.  Like so:

```javascript
  // on disconnect re-initialize pigpio
  pigpio.on('disconnected', (reason) => {
    console.log('rcencoder: received disconnect event')
    console.log('rcencoder: re-initializing pigpio')
    pigpio.connect()
    pigpio.once('connected', () => { init()})
  })
```

guymcswain, 2018-08-09:
Case 3 (dead connection) is now covered and tested (one time) successfully.   After I complete more testing I'll merge this branch into master.  Release to NPM as v1.1.0?

closed 2018-11-22 by guymcswain