guymcswain/pigpio-client #6: pigpiod on different port

karl0ss, 2019-04-29:
I am running pigpiod on a different port to the standard 8888, I am wanting to update my lights using pigs, but cant seem to work out how to get pigs to use the new port?

Thanks

guymcswain, 2019-04-29:
What does `pigs hwver` return?
Try `echo "hwver" > /dev/pigpio`
Both should give same result.

guymcswain, 2019-04-29:
Try `cat /dev/pigerr&` followed by above commands.

karl0ss, 2019-04-30:
Thanks for this, will have a play this evening when home from work, thank you for getting back to me.

karl0ss, 2019-04-30:
`pi@octopi:~ $ sudo pigpiod -p 8887
`
```
pi@octopi:~ $ sudo netstat -tulpn | grep :8887
tcp6       0      0 :::8887                 :::*                    LISTEN      1816/pigpiod
```
when I run 
`pigs hwver
`
or
`sudo pigs hwver
`

Terminal just hangs...

When I do -

`pi@octopi:~ $ sudo pigpiod
`
```
pi@octopi:~ $ sudo netstat -tulpn | grep :8888
tcp6       0      0 :::8888                 :::*                    LISTEN      1946/pigpiod
```

```
pi@octopi:~ $ pigs hwver
10494163
```

```
pi@octopi:~ $ cat /dev/pigerr&
[1] 1998
pi@octopi:~ $ echo "hwver" > /dev/pigpio
```

It seems that pigs just can't communicate with pigpiod when its not running on the standard port?

Thanks

guymcswain, 2019-04-30:
It seems so.  Hopefully Joan will respond to your question on the pigpio repo.

karl0ss, 2019-05-01:
Yeha, ill update my comment there with the findings here, I have changed my setup, so don't need to run off 8888, but I think it still something that needs looking at..

closed 2019-05-08 by guymcswain