guymcswain/pigpio-client #16: Network error not handled

louisvangeldrop, 2019-11-29:
MyError [pigpioClientError]: Unhandled socket error, connect ENETUNREACH 192.168.0.148:8888

guymcswain, 2019-11-29:
Generally means the address is not accessible.  Can you ping it?

louisvangeldrop, 2019-11-29:
It is my intention to disconnect my client from the pigpiod-server. A few minutes later I want to reconnect the client and start collecting my data from pigpiod. I noticed that a lot of errors are handled in pigpio-client.js, but this one isn't.

guymcswain, 2019-11-29:
Yes, the common connection errors are handled, and retried if you have enabled timeout, but other errors will be passed to the user application.

guymcswain, 2019-11-29:
I'm not a network expert but the host (pigpiod server) may require a timeout before allowing the same client to reconnect.  I have not tested intentionally disconnecting from the client then immediately reconnecting.

louisvangeldrop, 2019-12-02:
Neither am I. I use a Raspberry Pi Zero W as a rf433 transmitter and receiver. Another Raspberry Pi hosts a Node-red application, which will transmit/receive rf433 signals for further processing using pigpio-client. I want o be sure that if Raspberry Pi Zero W fails for whatever reason ( maybe switching off/on) the Node-red application reconnects. 
That is the reason that I disconnect and reconnect the Zero W

guymcswain, 2020-01-13:
Sorry, just getting back to this issue.

Pigpio-client will handle cases such as the daemon going away temporarily, as in the case of a reboot.  In this case EHOSTUNREACH or ECONNREFUSED are the usual suspects.  These will be retried until the 'timeout' option has expired and a ' disconnected' event will be generated. 
 However, ENETUNREACH implies the entire network interface is down.  (Presumably you are not restarting pigpio-client from a service before the network is available.).  Is this scenario likely to be encountered and how should the client handle it?  For now I believe the client propagates the error for the application to handle.

I need a bit more help/guidance from you on how to reproduce this problem.

louisvangeldrop, 2020-01-14:
I need some time in order to rebuild the same situation as a month ago. I have moved the client application to the same raspberrypi zero as the server ( running pigpiod).

louisvangeldrop, 2020-01-19:
In my case it happened, when my laptop goes into sleep-mode. When it wakes up, there is no wifi-connection with my router and the error occurs.

guymcswain, 2020-01-19:
Ok, that makes sense.  A use case I had not considered.  Thank you.

guymcswain, 2020-01-21:
Can you please enable debug messages and send to me the output captured at the time your laptop is resumed (when you get the ENETUNREACH error).  To enable the debug messages, set the environment variable "DEBUG" to 'pigpio'.  For example on command line do:

`DEBUG='pigpio' node myApp.js | grep "pigpio-client" > debug.log`

If you are running your node app from a service file, you'll need to modify the EXECSTART=.  Then 'grep' the output from journalctl:

`journalctl -u myApp --no-pager | grep "pigpio-client"

louisvangeldrop, 2020-02-03:
Took me a while to reproduce (holidays). See attached file for the output. I am using Windows 10 to test the program. Then I press the power key => hibernate. Restart the pc and it takes a minute to reconnect to the Access Point

louisvangeldrop, 2020-02-03:
Additonal error info from the command console:

events.js:200
      throw er; // Unhandled 'error' event
      ^
MyError [pigpioClientError]: Unhandled socket error, connect ENETUNREACH 192.168.0.148:8888
    at Socket.handler [as reconnectHandler] (C:\Users\louis\AppData\Roaming\npm\node_modules\pigpio-client\pigpio-client.js:219:28)
    at Socket.emit (events.js:223:5)
    at emitErrorNT (internal/streams/destroy.js:92:8)
    at emitErrorAndCloseNT (internal/streams/destroy.js:60:3)
    at processTicksAndRejections (internal/process/task_queues.js:81:21)
Emitted 'error' event on MyEmitter instance at:
    at Socket.handler [as reconnectHandler] (C:\Users\louis\AppData\Roaming\npm\node_modules\pigpio-client\pigpio-client.js:219:14)
    at Socket.emit (events.js:223:5)
    [... lines matching original stack trace ...]
    at processTicksAndRejections (internal/process/task_queues.js:81:21) {
  name: 'pigpioClientError',
  code: 'PI_CLIENT',
  message: 'Unhandled socket error, connect ENETUNREACH 192.168.0.148:8888',
  api: ''
}

Best regards,
Louis

guymcswain, 2020-02-05:
There is no attachment that I can see.  However, I've taken to try to reproduce this on my Windows 10 PC.  But without success so far.  Here is what I'm doing:

1. From Raspberry Pi host, mine is named 'pigpiov68', start pigpiod with:
pi@pigpiov68:~ $ `sudo pigpiod`

2. From Win10 PC, in a bash terminal (Windows sub-system for Linux), run  :
`guy@Amazigh:/mnt/d/Users/guy/Projects/pigpio-client$ DEBUG='pigpio' node ./test/connect.js`

> pigpio-client  commandSocket connected
pigpio-client  notificationSocket connected
pigpio-client  opened notification socket with handle= 1
pigpio-client  pigpio-client ready
Event=connected, arg=[object Object]
pigpio-client  bsc event monitoring active

3. Put PC to sleep

4. Wake up PC

> ...
pigpio-client  pigpio command socket end received
pigpio-client  pigpio notification socket end received
pigpio-client  notificationSocket closed
pigpio-client  notificationSocket destroyed due to closed unexpectedly
pigpio-client  commandSocket closed
pigpio-client  commandSocket destroyed due to closed unexpectedly
Event=disconnected, arg=closed unexpectedly
pigpio-client  sent disconnect event to application

The script, `./test/connect.js` contains:

```javascript
const connectOptions = {host: 'pigpiov68', timeout: 2};  // timeout!
const client = require('../pigpio-client.js');
const pigpio = client.pigpio(connectOptions); // start connecting

pigpio.on('connected', (arg) => {
  console.log(`Event=connected, arg=${arg}`);
});
pigpio.on('error', (arg) => {
  console.log(`Event=error, arg=${arg}`);
});
pigpio.on('disconnected', (arg) => {
  console.log(`Event=disconnected, arg=${arg}`);
});

process.on('SIGINT', handleSignal);
process.on('SIGTERM', handleSignal);
process.on('SIGCONT', handleSignal);
process.on('SIGTSTP', handleSignal);
process.on('SIGHUP', handleSignal);
process.on('SIGBREAK', handleSignal);
process.on('SIGPIPE', handleSignal);

function handleSignal(signal) {
  console.log(`Node process received ${signal}`);
}

// Ensure a way to exit process!
setTimeout( () => {
  console.log('Script timer has expired.  Exiting node process.');
  pigpio.end( () => {
    process.exit();
  });
}, 5*60*1000); // 5 minutes
```
So, from my setup, the sleep closes the connections to the pigpiod server and then connect.js receives a 'disconnected' event.  I never receive the error you are getting.  Can you help me match your configuration?

louisvangeldrop, 2020-02-05:
I have changed the the once(‘disconnected’) event into:

pigpio.on('disconnected', (arg) => {
  console.log(`Event=disconnected, arg=${arg}`);

  pigpio.connect()

});

The pigpio.connect() =>

pigpio-client  commandSocket error code: ENETUNREACH, message: connect ENETUNREACH 192.168.0.148:8888
Event=error, arg=pigpioClientError: Unhandled socket error, connect ENETUNREACH 192.168.0.148:8888
pigpio-client  notificationSocket error code: ENETUNREACH, message: connect ENETUNREACH 192.168.0.148:8888
Event=error, arg=pigpioClientError: Unhandled socket error, connect ENETUNREACH 192.168.0.148:8888

  1.  Put PC to sleep
  2.  Wake up PC

...

guymcswain, 2020-02-05:
Ok, thanks.  Now connect.js is getting this error:

> Event=error, arg=pigpioClientError: Unhandled socket error, getaddrinfo EAI_AGAIN pigpiov68 pigpiov68:8888

Then pigpio-client closes the socket  :(

> pigpio-client  commandSocket closed on error

guymcswain, 2020-02-05:
Google shows:

> EAI_AGAIN is a DNS lookup timed out error

So maybe I'm hitting a variant of the problem, which is, the network route has gone away.  I just don't know how many variants I need to account for: EHOSTUNREACH, ENETUNREACH, EAI_AGAIN,  other?

guymcswain, 2020-02-05:
I replaced 'pigpiov68' with its resolved address '192.168.100.6'.  Now it get 'ECONNREFUSED' as pigpio-client attempts to reconnect.  During the reconnect period connect.js also get error 'EPIPE' and the node process get 'SIGPIPE'.  Presumably this is due to attempting to write/read to a socket that is not ready.  But eventually, connection is re-established as expected.  But still no 'ENETUNREACH'.

Details below:

> pigpio-client  commandSocket connected
pigpio-client  notificationSocket connected
pigpio-client  opened notification socket with handle= 0
pigpio-client  pigpio-client ready
Event=connected, arg=[object Object]
pigpio-client  bsc event monitoring active
pigpio-client  pigpio command socket end received
pigpio-client  pigpio notification socket end received
pigpio-client  notificationSocket closed
pigpio-client  notificationSocket destroyed due to closed unexpectedly
pigpio-client  commandSocket closed
pigpio-client  commandSocket destroyed due to closed unexpectedly
Event=disconnected, arg=closed unexpectedly
pigpio-client  sent disconnect event to application
pigpio-client  commandSocket error code: ECONNREFUSED, message: connect ECONNREFUSED 192.168.100.6:8888
pigpio-client  retry connection on commandSocket in 5 sec ...
pigpio-client  notificationSocket error code: ECONNREFUSED, message: connect ECONNREFUSED 192.168.100.6:8888
pigpio-client  retry connection on notificationSocket in 5 sec ...
pigpio-client  notificationSocket closed on error
pigpio-client  commandSocket closed on error
pigpio-client  commandSocket error code: ECONNREFUSED, message: connect ECONNREFUSED 192.168.100.6:8888
pigpio-client  retry connection on commandSocket in 5 sec ...
pigpio-client  notificationSocket error code: ECONNREFUSED, message: connect ECONNREFUSED 192.168.100.6:8888
pigpio-client  retry connection on notificationSocket in 5 sec ...
pigpio-client  notificationSocket closed on error
pigpio-client  commandSocket closed on error
pigpio-client  commandSocket reconnected
pigpio-client  notificationSocket reconnected
pigpio-client  commandSocket error code: EPIPE, message: write EPIPE
Event=error, arg=pigpioClientError: Unhandled socket error, write EPIPE
Node process received SIGPIPE
Node process received SIGPIPE
pigpio-client  commandSocket closed on error
Unable to connect to pigpiod and no retry timeout option specified.  Process exiting.  Check the host address/port and if the daemon is running.
pigpio-client  opened notification socket with handle= 1
pigpio-client  pigpio notification socket end received
pigpio-client  notificationSocket closed
pigpio-client  notificationSocket destroyed due to closed unexpectedly
Event=disconnected, arg=closed unexpectedly
pigpio-client  sent disconnect event to application
pigpio-client  commandSocket reconnected
pigpio-client  notificationSocket reconnected
pigpio-client  opened notification socket with handle= 1
pigpio-client  pigpio-client ready
Event=connected, arg=[object Object]
pigpio-client  bsc event monitoring active

At this point all is well.   Then the script bails out because of the timer.

> Script timer has expired.  Exiting node process.
pigpio-client  pigpio command socket end received
pigpio-client  commandSocket closed
pigpio-client  commandSocket destroyed due to closed unexpectedly
pigpio-client  pigpio notification socket end received
pigpio-client  notificationSocket closed
pigpio-client  notificationSocket destroyed due to closed unexpectedly
Event=disconnected, arg=closed unexpectedly

louisvangeldrop, 2020-02-05:
I forgot to mention that I am using node directly from the command-prompt cmd. I also use an AP-repeater within the network.

guymcswain, 2020-02-06:
I'm going to just patch the code to handle both ENETUNREACH and EAI_AGAIN.  I'll push this up to master soon.  It will be rev 1.5.1.  I don't plan on doing a publish to npm since I'm planning 1.6.0 in the next week.

Toeedev, 2020-02-11:
So just to clarify, will there be a way to handle disconnection and try to reconnect?
I tried to run the code from above, and yes I get .on disconnect, but it never tried to reconnect I think. 

Also how to handle .write errors? 
```javascript
while (true) {
    await wait(2000)
    console.log("writing..")
    await led
      .write(1)
      .then((res) => {
        console.log(res)
      })
      .catch((err) => {
        console.log(err)
      })
  }
```

I have something like that, my idea is, when somehow it looses connection and tried to write, to error immediately, but instead, it just hangs for a minute with error: 
`Unable to connect to pigpiod and no retry timeout option specified.  Process exiting.  Check the host address/port and if the daemon is running.`
And after a minute it goes to `.on disconnect`

guymcswain, 2020-02-11:
@toeedev,
Your question should probably be opened as a new issue so as not to confuse with the ENETUNREACH after waking from sleep state.

Let me try to summarize what you are observing and what my recent testing has revealed:

When both sockets (command and notification) receive 'end/close' or 'error', the sockets are `destroy`ed and a 'disconnected' event is emitted.  The original intention was for the application to restart (by process.exit then systemd or forever restarts the app).  That is the safest way to handle such an event because you don't know if the pigpiod server was shutdown/rebooted and the internal states (such as waveform, pwm, etc) was cleared.  This is how I'm using the 'disconnected' event in my home automation and it seems very stable for my needs.  

Pigpio-client also provides a method to reconnect using the `connect` method.  My recent testing reveals that sockets that have been destroyed should not be re-used/re-connected but instead new sockets should be created.  This is a bug and I'm currently working on a solution.  I would be very happy to have you test it when it is ready - in the next couple of days.

Toeedev, 2020-02-11:
I will definitely be happy to test it. 
I don't want to close my script and reopen it. Program I want to write is, I have 2 devices local and remote.  it does same things on both, but if remote is gone for any reason, just keep it doing on local, and when remote is back keep it doing there as well..I don't mind how it errors, as long as it doesn't stop whole thing from running and can reconnect later.

guymcswain, 2020-02-12:
I pushed a branch named 'issue16' that should fix the ENETUNREACH as well as create new sockets when `pigpio.connect()` method is used to reconnect.  Please `git clone` this branch and test it in your applications then let me know if I should release the change.  Thanks in advance!

louisvangeldrop, 2020-02-12:
Got this error:

  '0': 'c',
  '1': 'o',
  '2': 'm',
  '3': 'm',
  '4': 'a',
  '5': 'n',
  '6': 'd',
  '7': 'S',
  '8': 'o',
  '9': 'c',
  '10': 'k',
  '11': 'e',
  '12': 't',
  '13': ' ',
  '14': 'E',
  '15': 'C',
  '16': 'O',
  '17': 'N',
  '18': 'N',
  '19': 'R',
  '20': 'E',
  '21': 'S',
  '22': 'E',
  '23': 'T',
  '24': ',',
  '25': ' ',
  '26': 'd',
  '27': 'i',
  '28': 's',
  '29': 'c',
  '30': 'o',
  '31': 'n',
  '32': 'n',
  '33': 'e',
  '34': 'c',
  '35': 't',
  '36': 'i',
  '37': 'n',
  '38': 'g',
  datum: '2020-2-12 12:41:53'
}

events.js:200
      throw er; // Unhandled 'error' event
      ^
MyError [pigpioClientError]: Unhandled socket error, connect ENETUNREACH 192.168.0.148:8888
    at Socket.handler [as reconnectHandler] (C:\Users\louis\OneDrive\Apps\Repos\IoT\Raspberry\NodeJS\rf433\node_modules\pigpio-client\pigpio-client.js:219:28)
    at Socket.emit (events.js:223:5)
    at emitErrorNT (internal/streams/destroy.js:92:8)
    at emitErrorAndCloseNT (internal/streams/destroy.js:60:3)
    at processTicksAndRejections (internal/process/task_queues.js:81:21)
Emitted 'error' event on MyEmitter instance at:
    at Socket.handler [as reconnectHandler] (C:\Users\louis\OneDrive\Apps\Repos\IoT\Raspberry\NodeJS\rf433\node_modules\pigpio-client\pigpio-client.js:219:14)
    at Socket.emit (events.js:223:5)
    [... lines matching original stack trace ...]
    at processTicksAndRejections (internal/process/task_queues.js:81:21) {
  name: 'pigpioClientError',
  code: 'PI_CLIENT',
  message: 'Unhandled socket error, connect ENETUNREACH 192.168.0.148:8888',
  api: ''

uneasymusic, 2020-02-12:
Can you please help m understand how it supposed to work? I don't really understand timeout I think.
I am connecting fine and getting 'connected' event. 
However it keeps triggering 'disconnected' and 'connected' frequently. For no reason really. 

- when I disconnect device from network, I get disconnected event. After a while of disconnection, I get error `Unhandled socket error, connect ETIMEDOUT 43.210.8.134:8888` 
and It keeps trying to reconnect ( as per my code). 
When I connect device back to netwrok, I get  couple of 'connected' events. 
So does that mean I have couple sockets opened? 

Also while I am doing `led.write` and disonnect device, it will timeout eventually but it doesn't stop the code and after connection back, it keeps doing writing. And that's briliant. 

My only concern is, if I have loads of those disconnections and connections, doesn't that use resources?

Good thing is, it doesn't crash  and reconnects when device is back in network. 
```javascript
const connectOptions = { host: '43.210.8.134', timeout: 1 } // timeout!
const client = require('pigpio-client')
const pigpio = client.pigpio(connectOptions) // start connecting

pigpio.on('connected', (arg) => {
  console.log(`Event=connected, arg=${arg}`)
})
pigpio.on('error', (arg) => {
  console.log(`Event=error, arg=${arg}`)
  pigpio.connect()
})
pigpio.on('disconnected', (arg) => {
  console.log(`Event=disconnected, arg=${arg}`)
  pigpio.connect()
})
```

guymcswain, 2020-02-12:
@louisvangeldrop ,
You also need an error handler.  Try:
```javascript
pigpio.on('error', (arg) => {
  console.log(`Event=error, arg=${arg}`);
  pigpio.connect()
});
```
Also, if you could turn on debug messages that would be helpful:
`DEBUG='pigpio' node <your script name>`

guymcswain, 2020-02-12:
@uneasymusic 
> However it keeps triggering 'disconnected' and 'connected' frequently.

Try timeout value of 2.  I'll explain later.

guymcswain, 2020-02-12:
Also, the retry timer (connection has not yet been established) is set to 5  seconds.  So you should see reconnect attempts every 5 seconds but I do not expect 'connected' / 'disconnected' events while this is occurring.

guymcswain, 2020-02-12:
> When I connect device back to netwrok, I get couple of 'connected' events.

I think the problem is in calling pigpio.connect() in both the 'error' handler and the 'disconnected' handler.  A socket error is followed by a socket 'close' event which is the actual trigger for 'disconnected'.  So please change your error handlers to just this:
```javascript
pigpio.on('error', (arg) => {
  console.log(`Event=error, arg=${arg}`);
});

Toeedev, 2020-02-12:
(uneasymusic it's me, just keep forgetting to logout)
 
The issue with above solution is, if connection times out and will never try again.  So if device will come back after timeout, it won't be picked up again.

guymcswain, 2020-02-12:
Meaning, if the keep-alive times out you don't receive a 'disconnected' event?  If so that's a bug I'll need to fix.  I'll have more time to test later this evening.  Thanks for the feedback.

Toeedev, 2020-02-12:
Well, here is my scenario. When I am testing it, just connection, without doing any led.write etc. 
I get disconnected, and because I have pigpio.connect() eventually  I get timeout error. 
And Because I have pigpio.connect() on error too, it keeps timing out. When I get device back to network, I get connected. 

```
Running...
Event=connected, arg=[object Object]  
//device disconnected and after around 30seconds
Event=disconnected, arg=timeout
//device still disconnected
Event=error, arg=pigpioClientError: Unhandled socket error, connect ETIMEDOUT 192.168.1.106:8888
// still 
Event=error, arg=pigpioClientError: Unhandled socket error, connect ETIMEDOUT 192.168.1.106:8888
//still
Event=error, arg=pigpioClientError: Unhandled socket error, connect ETIMEDOUT 192.168.1.106:8888
// device is connected back and after some seconds: 
Event=connected, arg=[object Object]
```

It's different when I am writing to device and it gets disconnected then. 
I'am not getting anything until I get device back to network, I then get 2x 'disconnected'

guymcswain, 2020-02-12:
This is good information.  I'm using a proxy to disconnect.  How are you creating the disconnect?

Toeedev, 2020-02-12:
Well, I either unplug Ethernet from raspberry, or I am unpluging the power ;)

guymcswain, 2020-02-12:
> I either unplug Ethernet from raspberry,

This scenario should be similar to my proxy when I pause the sockets.  In my testing, I get the

> pigpio-client  Pigpio keep-alive packet not received before timeout expired

 debug message, then receive 'disconnected' event, then call `pigpio.connect()` and receive the 'connected' event - because my proxy allows new connections to be formed.  I can prevent this and rerun my testing using the proxy.

> I am unpluging the power

In this scenario, a RST should be received on the socket.  I don't know how to simulate this with my proxy.  I was hoping to avoid it but I'll try a remote shutdown and pull the power then power back up.

guymcswain, 2020-02-12:
If anyone is interested, the proxy I'm using can be found [here](https://gist.github.com/guymcswain/e1c710b7719ccde518c0fc2bab7bc935).

louisvangeldrop, 2020-02-13:
I do not the ENETUNREACH error anymore.

However if start my app, while my pc is ENETUNREACH-status, I get the following error:

MyError [pigpioClientError]: Unhandled socket error, connect ENETUNREACH 192.168.0.148:8888
    at Socket.handler [as reconnectHandler] (C:\Users\louis\OneDrive\Apps\Repos\IoT\Raspberry\NodeJS\rf433\node_modules\pigpio-client\pigpio-client.js:219:28)
    at Socket.emit (events.js:223:5)
    at emitErrorNT (internal/streams/destroy.js:92:8)
    at emitErrorAndCloseNT (internal/streams/destroy.js:60:3)
    at processTicksAndRejections (internal/process/task_queues.js:81:21)
Emitted 'error' event on MyEmitter instance at:
    at Socket.handler [as reconnectHandler] (C:\Users\louis\OneDrive\Apps\Repos\IoT\Raspberry\NodeJS\rf433\node_modules\pigpio-client\pigpio-client.js:219:14)
    at Socket.emit (events.js:223:5)
    [... lines matching original stack trace ...]
    at processTicksAndRejections (internal/process/task_queues.js:81:21) {
  name: 'pigpioClientError',
  code: 'PI_CLIENT',
  message: 'Unhandled socket error, connect ENETUNREACH 192.168.0.148:8888',
  api: ''
}

The socket is not yet in the connecting state.

guymcswain, 2020-02-13:
@louisvangeldrop , thanks for the feedback.

>   However if start my app, while my pc is ENETUNREACH-status, I get the following error ,,,

I understand why.  It is because only certain errors are qualified with the connecting state.  I will fix this.

@Toeedev , ETIMEDOUT is another type of network error that I currently don't check for.  I'm going to change the retry to be on any network error.  It appears that each OS, environment can have a different 'error signature'.  I will fix this.

guymcswain, 2020-02-13:
Thank you for your patience.  I have pushed a change which retries on **any** socket error.  I tested with both the pigpiod proxy as well as a live power pull test on the raspberry.  Please let me know when you get some time to run your testing.

Toeedev, 2020-02-14:
ok so I removed `pigpio.connect()` from on.error. 
I disconnected PI ( pulled ethernet out)  and I get that: 

```
Running...
Event=connected, arg=[object Object]
Event=disconnected, arg=timeout
ETIMEDOUT, retrying 43.210.8.134:8888 ...
Event=error, arg=pigpioClientError: Could not connect, retry timeout expired
Unable to connect to pigpiod.  No retry timeout option was specified.  Verify that daemon is running from 43.210.8.134:8888.
Event=error, arg=pigpioClientError: connect ETIMEDOUT 43.210.8.134:8888
```

That is after around 2 minuets, and then I plug Pi back, and I get nothing else after that. 
No 'connected', so I guess it stays disconnected. 
Is there any chance to have infinite retry?

guymcswain, 2020-02-14:
The timeout should have reinitialized after you call `pigpio.connect`.  By calling that API from your app you essentially create the infinite retry.  I'll verify this test case and fix.  Thanks!

guymcswain, 2020-02-15:
I pushed an update to branch 'issue16'.

When a connection retry timeout occurs, an 'error' event will be generated (and not a 'disconnected' event due to the fact the current state is already disconnected).  Therefore, in this case, you will need to monitor the 'error' event and initiate another `pigpio.connect()` to continue the retry process.  I have improved the error object that is returned to include an api property that will allow the application to easily filter for this condition - as seen in the following example:
```javascript
pigpio.on('error', e => {
  // continue retry indefinitely
  if (e.code === 'PI_CLIENT' && e.api === 'connect') {
    pigpio.connect();
  }
});

Toeedev, 2020-02-17:
That works great. So now there is only one last thing left for me ;). 
Is it possible to make it more frequent ? Like now timeout is set to '2', but that seems long to discover disconnection and connection after. 
0.3 seems like nice value, but then it goes crazy, when connected it keeps  disconnecting with 
`Event=disconnected, arg=pigpio keep-alive timeout` every couple of seconds. 
And the when disconnected, I only get disconnected event, and nothing after, but when I connecting it back, it connects just fine.

guymcswain, 2020-02-17:
@Toeedev , I'm happy the 'pigpio keep-alive timeout' and subsequent re-connection scheme is working for you.  Thanks for your testing that exposed problems in this area!

> Is it possible to make it more frequent ? Like now timeout is set to '2', but that seems long to discover disconnection and connection after.

As an enhancement to the library, yes.  Please open a separate issue on that subject (as this one has become epic!).

@louisvangeldrop , have you had a chance to test the latest?

louisvangeldrop, 2020-02-17:
I will test it Wednesday. I am out of home for a couple of days

louisvangeldrop, 2020-02-20:
I did a test this morning. I can’t reproduce the ENETUNREACH error, but I got this error:

pigpio-client  commandSocket connected
pigpio-client  notificationSocket connected
pigpio-client  opened notification socket with handle= 1
pigpio-client  pigpio-client ready
pigpio-client  bsc event monitoring active
pigpio-client  notificationSocket error code: ECONNRESET, message: read ECONNRESET
pigpio-client  notificationSocket destroyed due to notificationSocket ECONNRESET, disconnecting
pigpio-client  notificationSocket closed on error
pigpio-client  commandSocket error code: ECONNRESET, message: read ECONNRESET
pigpio-client  commandSocket destroyed due to commandSocket ECONNRESET, disconnecting
Event=disconnected, arg=commandSocket ECONNRESET, disconnecting
pigpio-client  sent disconnect event to application
pigpio-client  commandSocket closed on error
Unable to connect to pigpiod and no retry timeout option specified.  Process exiting.  Check the host address/port and if the daemon is running.

guymcswain, 2020-02-20:
Please help me to interpret what events are occurring when examining this log:

You started your application here?

> pigpio-client  commandSocket connected
pigpio-client  notificationSocket connected
pigpio-client  opened notification socket with handle= 1
pigpio-client  pigpio-client ready
pigpio-client  bsc event monitoring active

You place your laptop into sleep mode, then later wake it up here?

> pigpio-client  notificationSocket error code: ECONNRESET, message: read ECONNRESET
pigpio-client  notificationSocket destroyed due to notificationSocket ECONNRESET, disconnecting
pigpio-client  notificationSocket closed on error
pigpio-client  commandSocket error code: ECONNRESET, message: read ECONNRESET
pigpio-client  commandSocket destroyed due to commandSocket ECONNRESET, disconnecting
Event=disconnected, arg=commandSocket ECONNRESET, disconnecting
pigpio-client  sent disconnect event to application
pigpio-client  commandSocket closed on error
Unable to connect to pigpiod and no retry timeout option specified.  Process exiting.  Check the host address/port and if the daemon is running.

The very last message:

> Unable to connect to pigpiod and no retry timeout option specified.  Process exiting.  Check the host address/port and if the daemon is running.

looks like a faulty message that I corrected in my last update.  Please make sure you got the latest commit `b3449f6` from branch 'issue16'.  

Also, it would be helpful if I could see a log message from your application when it receives/handles the 'connected', 'disconnected' and 'error' events from the pigpio object.

louisvangeldrop, 2020-02-20:
My apologies, I have cloned the wrong branch.
-Louis

louisvangeldrop, 2020-02-20:
I used the latest version still this message when I disconnect Wifi connection:

[cid:image002.jpg@01D5E830.13875C50]

louisvangeldrop, 2020-02-20:
Used code:

// import * as piGpio from "../node_modules/pigpio-client/pigpio-client.js"
export var piGpioClient
export var piGpio

export var HIGH = 1;
export var LOW = 0;
export var OUTPUT = 'output';
export var INPUT = 'input';

interface options {
  host?: string,
  port?: number,
  /**
       * Timeout in minutes
       */
  timeout?: number,
  /**
   * INPUT, OUTPUT, ALT0, ALT1, ALT2, ALT3, ALT4, or ALT5 (optional, no default)
   */
  mode?: string,

  /**
   * PUD_OFF, PUD_DOWN, or PUD_UP (optional, no default)
   */
  pullUpDown?: number,

  /**
   * interrupt edge for inputs. RISING_EDGE, FALLING_EDGE, or EITHER_EDGE (optional, no default)
   */
  edge?: number,

  /**
   * interrupt timeout in milliseconds (optional, defaults to 0 meaning no timeout if edge specified)
   */

  /**
   * boolean specifying whether or not alert events are emitted when the GPIO changes state (optional, default false)
   */
  alert?: false,

}

export var connect = (options?: options) => {
  return new Promise((resolve, reject) => {
    piGpioClient = require('../node_modules/pigpio-client/pigpio-client')//.pigpio({ host: host })
    piGpio = piGpioClient.pigpio(options)

    piGpio.once('connected', resolve)

    piGpio.once('error', err => {
      console.log(`Event=error, arg=${err}`);
      // continue retry indefinitely
      if (err.code === 'PI_CLIENT' && err.api === 'connect') {
        resolve(piGpio.connect)
      }

    })

    piGpio.once('disconnected', err => {
      console.log(`Event=disconnected, arg=${err}`);
      // console.log({ datum: new Date().toLocaleString(), ...err })
      reject(piGpio.connect) //????????
    })

  });
}

guymcswain, 2020-02-21:
```javascript
// your module
export ...

export var connect = (options?: options) => {
  return new Promise((resolve, reject) => {
    piGpioClient = require('../node_modules/pigpio-client/pigpio-client')//.pigpio({ host: host })
    piGpio = piGpioClient.pigpio(options);

    piGpio.once('connected', () => {
      piGpio.removeAllListeners();
      resolve();
    });

    // We are starting the app so the network is not sleeping and should
    // connect right away, or at least within timeout (minimum 2 minutes).
    piGpio.on('error', err => {
      console.log(`Event=error, arg=${err}`);
      if (err.message.includes('timeout')) {
        reject('Awake but network is not available!?');
      }
    });
    // We only care about 'disconnect' in the context of the main app.
    //piGpio.once('disconnected', err => {
    //  console.log(`Event=disconnected, arg=${err}`);
    //  // console.log({ datum: new Date().toLocaleString(), ...err })
    //  reject(piGpio.connect) //????????
    //})

  });
}

// your app starts here
connect().then( () => {
    // Handle more than a single disconnect.
    piGpio.on('disconnected', err => {
      console.log(`Event=disconnected, arg=${err}`);
      piGpio.connect();  // Reconnect to read sensors.  Don't care about pigpiod state!
    })
  // initialize sensors
  // read sensors
}).catch(console.log);

// You need an error handler in the main app.
piGpio.on('error', err => {
  console.log(`Event=error, arg=${err}`);
})

guymcswain, 2020-02-21:
Please reply using the GitHub issues dialogue web page.  It avoids all the email headers and body and allows a cleaner view of the code when you use appropriate markdown.  Thanks.

louisvangeldrop, 2020-02-21:
Sorry, you are right,

I will reply next time using Github issues.

louisvangeldrop, 2020-02-21:
If I disconnect the Wifi AP the following error occurs even if I have specified a timeout:

pigpio-client  commandSocket error code: ECONNRESET, message: read ECONNRESET
Unable to connect to pigpiod.  No retry timeout option was specified.  Verify that daemon is running from 192.168.0.148:8888. pigpio-client.js:224
MyError

guymcswain, 2020-02-21:
Just for sanity, can you change this

```javascript
piGpio.once('connected', () => {
  piGpio.removeAllListeners();
  resolve();
});
```
To this
```javascript
piGpio.once('connected', (info) => {
  console.log(JSON.stringify(info,null,2));
  piGpio.removeAllListeners();
  resolve();
});
```
... and report what the info object contains?

louisvangeldrop, 2020-02-21:
Ok. I got it fixed within my module calling exports.gpio.connect()

guymcswain, 2020-02-21:
That is good to know.  Thanks for your help in getting this issue resolved!

PR #24 (2020-02-23): Retry connection on all network errors.
Fixes #16.
closed 2020-02-23 by guymcswain