guymcswain/pigpio-client #17: Error using waveform

louisvangeldrop, 2019-12-04:
The following code always returns a waveform error on my Raspberry Pi Zero W:

      let wc = await this.gpio.waveClear()
      let wap = await this.gpio.waveAddPulse(pulses)
      var wid = await this.gpio.waveCreate()
      let wso = await this.gpio.waveSendOnce(wid)
      let wdel = await this.gpio.waveDelete(wid)

Pulses is array of 132 tuples.

MyError [pigpioError]: attempt to create an empty waveform
    at responseHandler (C:\Users\louis\AppData\Roaming\npm\node_modules\pigpio-client\pigpio-client.js:293:17)
    at Socket.<anonymous> (C:\Users\louis\AppData\Roaming\npm\node_modules\pigpio-client\pigpio-client.js:343:30)
    at Socket.emit (events.js:210:5)
    at addChunk (_stream_readable.js:326:12)
    at readableAddChunk (_stream_readable.js:301:11)
    at Socket.Readable.push (_stream_readable.js:235:10)
    at TCP.onStreamRead (internal/stream_base_commons.js:182:23) {
  name: 'pigpioError',
  code: 'PI_EMPTY_WAVEFORM',
  message: 'attempt to create an empty waveform',
  api: 'WVCRE'
}

guymcswain, 2019-12-04:
Hmmm.  That sequence of code should work except for a potential problem with deleting the waveform before transmission is complete.  However, in this case it fails on `waveCreate()`.

Some questions:
1)  Are you sure pulses contain non-zero delay values?
2)  Are you creating multiple instances of pigpio-client object?  `this.gpio.*`

louisvangeldrop, 2019-12-04:
1. I have just checked the pulses array and it does not contain zero values
2. Only one pigpio-client object is running.
I assume that the pulses are in microseconds.

louisvangeldrop, 2019-12-04:
JSON.stringify(pulses)
"[[1,0,260],[0,1,2730],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,10400]]"

guymcswain, 2019-12-04:
Correct, delays are in microseconds.

And nothing else is connected/running on pigpiod?

What are the return values?
`wc` (expect 0)
`wap` (expect 132)
`wid` (expect 0)

louisvangeldrop, 2019-12-05:
Yes, that are the return codes

guymcswain, 2019-12-05:
Ok, I'll run a test tonight on my zero W.  What version of pigpio and pigpio-client are you running?

louisvangeldrop, 2019-12-05:
I run pigpio-client@1.4.0 at a Windows 10 pc.
pigpiod -v => 64

louisvangeldrop, 2019-12-05:
Installed pigpiod 72. No difference

guymcswain, 2019-12-06:
I'm not able to reproduce the error in question.  I'm using pigpio V68 and pigpio-client v1.4 (but info object shows 1.3).

The script:
```javascript
const pigpio = require('./pigpio-client.js').pigpio({host: '192.168.100.6'});  

const ready = new Promise((resolve, reject) => {
  pigpio.once('connected', resolve);
  pigpio.once('error', reject);
});

function wait(ms) {
  return new Promise(resolve => setTimeout(resolve, ms));
}

ready.then(async (info) => {
  // display information on pigpio and connection status
  console.log(JSON.stringify(info,null,2));

  // Test waves output to LED on GPIO 4
  const led = pigpio.gpio(4);
  await led.modeSet('output');
  
  // report events on led
  var lastTick;
  led.notify((level, tick) => {
    if (lastTick)
      console.log(`LED changed to ${level} after ${tick-lastTick} usec`);
    lastTick = tick;
  });
  
  let pulses = [[1,0,260],[0,1,2730],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,260],[1,0,260],[0,1,1300],[1,0,260],[0,1,10400]];
  
  // use waves to blink the LED rapidly (toggle every 100ms)
  let wc = await led.waveClear();
  let wap = await led.waveAddPulse(pulses);
  let wid = await led.waveCreate();
  let wso = await led.waveSendOnce(wid);
  let wdel = await led.waveDelete(wid);
  
  console.log('Done');

}).catch(console.error);
```

The output:

> LED changed to 0 after 260 usec
> Done
> LED changed to 1 after 2730 usec
> LED changed to 0 after 260 usec
> LED changed to 1 after 260 usec
> LED changed to 0 after 260 usec
> LED changed to 1 after 1300 usec
> LED changed to 0 after 259 usec
> LED changed to 1 after 260 usec
> LED changed to 0 after 261 usec
> LED changed to 1 after 1300 usec
> LED changed to 0 after 260 usec
> LED changed to 1 after 1300 usec
> LED changed to 0 after 260 usec
> LED changed to 1 after 260 usec
> LED changed to 0 after 260 usec
> LED changed to 1 after 260 usec
> LED changed to 0 after 260 usec
> LED changed to 1 after 1300 usec
> LED changed to 0 after 260 usec
> LED changed to 1 after 260 usec
> LED changed to 0 after 260 usec
> LED changed to 1 after 1300 usec
[... 84 lines cut ...]
> LED changed to 0 after 260 usec
> LED changed to 1 after 260 usec
> LED changed to 0 after 260 usec
> LED changed to 1 after 1300 usec
> LED changed to 0 after 260 usec
> LED changed to 1 after 260 usec
> LED changed to 0 after 260 usec
> LED changed to 1 after 1300 usec
> LED changed to 0 after 260 usec
> LED changed to 1 after 260 usec
> LED changed to 0 after 260 usec
> LED changed to 1 after 1300 usec
> LED changed to 0 after 260 usec
> LED changed to 1 after 260 usec
> LED changed to 0 after 260 usec
> LED changed to 1 after 1300 usec
> LED changed to 0 after 260 usec
> LED changed to 1 after 260 usec
> LED changed to 0 after 260 usec
> LED changed to 1 after 1300 usec
> LED changed to 0 after 260 usec
> LED changed to 1 after 260 usec
> LED changed to 0 after 260 usec
> LED changed to 1 after 1300 usec
> LED changed to 0 after 260 usec

louisvangeldrop, 2019-12-06:
I have copied your javascript-file. Also here no error. I have noticed that an wavecreate error occurs with missing "await" statement. The function that calls the waveform.create etc  is called from http.createserver event. I have to figure out how to make this async/await aware.

louisvangeldrop, 2019-12-06:
Thanks for the proper help.

closed 2019-12-10 by guymcswain