guymcswain/pigpio-client #18: I2C support

bozonx, 2019-12-16:
It would be great to have I2C support. For example to connect to PCF8574 port expander or I2C sensors.

guymcswain, 2019-12-16:
I can provide support for the APIs you require but I don't have I2C hardware to test against.  If you are willing to test and provide feedback then I will help you.

First, which I2C APIs do you need?  There are many:

> i2c_open | Opens an I2C device
> -- | --
> i2c_close | Closes an I2C device
>   |  
> i2c_write_quick | smbus write quick
>   |  
> i2c_read_byte | smbus read byte
> i2c_write_byte | smbus write byte
>   |  
> i2c_read_byte_data | smbus read byte data
> i2c_write_byte_data | smbus write byte data
>   |  
> i2c_read_word_data | smbus read word data
> i2c_write_word_data | smbus write word data
>   |  
> i2c_read_block_data | smbus read block data
> i2c_write_block_data | smbus write block data
>   |  
> i2c_read_i2c_block_data | smbus read I2C block data
> i2c_write_i2c_block_data | smbus write I2C block data
>   |  
> i2c_read_device | Reads the raw I2C device
> i2c_write_device | Writes the raw I2C device
>   |  
> i2c_process_call | smbus process call
> i2c_block_process_call | smbus block process call
>   |  
> i2c_zip | Performs multiple I2C transactions
> 
>

bozonx, 2019-12-17:
I'm glad to help the project, I'll test I2C functionality on my devices.
For the first time these functions are enough: i2cOpen, i2cClose, i2cReadDevice, i2cWriteDevice. I watched http://abyz.me.uk/rpi/pigpio/cif.html

guymcswain, 2019-12-17:
Ok, I have written some wrappers for these APIs for you to test.  [Here is a link to the gist with the wrappers.](https://gist.github.com/guymcswain/c0f34c4cd6a38bfc2fd3108a1da3ae8d)

guymcswain, 2019-12-17:
Sorry, those wrappers will not work unless I add the new commands using extended data arguments.  I will create a branch called 'i2c' and make this modification.

bozonx, 2019-12-19:
I tried that gist and had the result:

* i2cOpen works properly and returns an id of connection.
* i2cWriteDevice - does nothing, promise wasn't fulfilled at all.
* i2cReadDevice - throws an error

```
MyError [pigpioError]: unknown command
    at responseHandler (/home/ivan/workspace/squidlet/build/pigpio-client/pigpio-client.js:294:17)
    at Socket.<anonymous> (/home/ivan/workspace/squidlet/build/pigpio-client/pigpio-client.js:344:30)
    at Socket.emit (events.js:210:5)
    at Socket.EventEmitter.emit (domain.js:478:20)
    at addChunk (_stream_readable.js:326:12)
    at readableAddChunk (_stream_readable.js:301:11)
    at Socket.Readable.push (_stream_readable.js:235:10)
    at TCP.onStreamRead (internal/stream_base_commons.js:182:23) {
  name: 'pigpioError',
  code: 'PI_UNKNOWN_COMMAND',
  message: 'unknown command',
  api: ''
```

* i2cClose - throws the same error if pass a callback

guymcswain, 2019-12-19:
Did you download the 'i2c' branch?  It looks like the commands are missing.

In my testing it hangs on the i2cOpen(), presumably because I don't have I2C hardware attached to my RPi.  When I use a dummy device handler on i2cWrite(), I get:

> { pigpioError: unknown handle
    at responseHandler (/home/pi/i2c/pigpio-client/pigpio-client.js:294:17)
    at Socket.commandSocket.on (/home/pi/i2c/pigpio-client/pigpio-client.js:344:30)
    at Socket.emit (events.js:198:13)
    at addChunk (_stream_readable.js:288:12)
    at readableAddChunk (_stream_readable.js:269:11)
    at Socket.Readable.push (_stream_readable.js:224:10)
    at TCP.onStreamRead [as onread] (internal/stream_base_commons.js:94:17)
  name: 'pigpioError',
  code: 'PI_BAD_HANDLE',
  message: 'unknown handle',
  api: 'I2CWD' }

I can't really do much more without an I2C device attached.

guymcswain, 2019-12-19:
I have edited the gist.  There was an error in ic2ReadDevice().   Please try the new version and make sure you are using the i2c branch of pigpio-client repo.

bozonx, 2019-12-20:
I cloned the repo and checked out to i2c branch, i checked that `I2CO` etc exist in `pigpio-client.js`.
But at the moment it even doesn't fulfill the i2cOpen promise. But if set 0 instead of 4 in command `pi.request(I2CO, bus, device, 0, callback, flags)` it works and returns a number. Other commands don't work - don't fulfill their promises.

guymcswain, 2019-12-20:
I will need to find an I2C device to attach to my raspberry pi in order to debug this.  Until then, could you run and collect some debug information for me?  To do this, set PIGPIO environment variable to 1 (or true) as such:
`pi@hostname:~ $  PIGPIO=1 node i2cScript.js`

This will output messages to the console for each request and response on the socket interface.

Thank you for your assistance.

guymcswain, 2019-12-21:
I was able to use the BSC peripheral as an i2c slave device for testing.  I found errors in several of the APIs.  Please try revision 5 of the i2cWrappers.js gist.

bozonx, 2019-12-23:
I tested the new gist and there are i2cOpen, i2cClose and i2cWriteDevice functions work properly. But i2cReadDevice always returns `[0]` as data.

there is a debug info (I only changed `const I2C_Message = [1];`):

```
request=  26 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
response=  26 0 0 0 0 0 0 0 0 0 0 0 64 0 0 0
request=  17 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
response=  17 0 0 0 0 0 0 0 0 0 0 0 130 32 162 0
request=  54 0 0 0 1 0 0 0 34 0 0 0 4 0 0 0
extended params=  0 0 0 0
response=  54 0 0 0 1 0 0 0 34 0 0 0 19 0 0 0
request=  57 0 0 0 19 0 0 0 0 0 0 0 1 0 0 0
extended params=  1
response=  57 0 0 0 19 0 0 0 0 0 0 0 0 0 0 0
request=  56 0 0 0 19 0 0 0 1 0 0 0 0 0 0 0
response=  56 0 0 0 19 0 0 0 1 0 0 0 1 0 0 0
extended params=  0
received 1 bytes:  [ 0 ]
```

guymcswain, 2019-12-23:
Yes, I had to convert the string to a buffer.  Try: `Buffer.from(I2C_Message, 'utf8')` as the argument for write.  Eventually, I will clean up the API to accept string or buffer.

bozonx, 2019-12-23:
The `i2cWriteDevice` works with a buffer and Uint8Array as an argument. But problem with `i2cReadDevice` it always returns `[ 0 ]`

guymcswain, 2019-12-23:
This works for me:
```javascript
const I2C_Message = "Hello I2C";
ready.then(async () => {
  let device = await i2cOpen(1, 0x13);
  console.log('Handle = ', device);
  let buf = Buffer.from(I2C_Message, 'utf8');
  console.log('Sending: ', buf);
  await i2cWriteDevice(device, buf);
  await new Promise( (resolve, reject) => setTimeout(resolve, 1000));
  let [count, ...data] = await i2cReadDevice(device, I2C_Message.length);
  console.log(`received ${count} bytes: `)
  console.log('message = ', data.map(x=> String.fromCodePoint(x)).join(''));
  await i2cClose(device);
  pi.end();
}).catch(console.error);
```

guymcswain, 2019-12-23:
For completeness, I'm controlling the BSC peripheral using this Python script.  Run it in a separate terminal.

```Python
#!/usr/bin/env python
import time
import pigpio

I2C_ADDR=0x13

def i2c(id, tick):
    global pi

    s, b, d = pi.bsc_i2c(I2C_ADDR)
    if b:
        print("received {}".format(d))
        s, b, d = pi.bsc_i2c(I2C_ADDR, d)

pi = pigpio.pi()

if not pi.connected:
    exit()

# Respond to BSC slave activity

e = pi.event_callback(pigpio.EVENT_BSC, i2c)

pi.bsc_i2c(I2C_ADDR) # Configure BSC as I2C slave

time.sleep(600)

e.cancel()

pi.bsc_i2c(0) # Disable BSC peripheral

pi.stop()
```
The script simply echos whatever it receives.

guymcswain, 2019-12-23:
... and you need to connect GPIO 2 to GPIO 18 and GPIO3 to GPIO19.

bozonx, 2019-12-24:
It actually works. Something was wrong with one of my pcf8574 board. Thanks for this feature and sorry that I respond rarely.

guymcswain, 2019-12-24:
Thanks for testing.  I will integrate these APIs into the I2C branch later this week.  I'm going to make modifications to allow string or buffer arguments.  I will let you know when it is ready.  I value getting your feedback on the final version before I merge into the master release.

bozonx, 2019-12-24:
I'll check it.
And I have some more question: Is there ability to check is I2C bus opened or not on a server side (pigpio daemon)? I would like to check that id which `i2cOpen` returns.

guymcswain, 2019-12-24:
I don't think it is possible to know the id (handle) before invoking `i2cOpen`.  If the Id returned is 0, you can infer that you are the first and only client requesting this I2C bus.  But I'm not sure if I understand what you want to do.

bozonx, 2019-12-24:
I want to check the situation when I've got a handle but the pigpiod has been restarted and doesn't remember this handle. But I realized that reading and writing with wrong handle throw an error:
```
{
  name: 'pigpioError',
  code: 'PI_BAD_HANDLE',
  message: 'unknown handle',
  api: 'I2CWD'
}
```
And eventually I think it's enough to handle this situation.

bozonx, 2020-01-21:
Could you publish these changes to NPM? Thanks in advance.

guymcswain, 2020-01-21:
Ok, thanks for the reminder.  I've been busy with pigpio repo but I will get back to pigpio-client and do this in the next few days.

guymcswain, 2020-01-23:
Done.  I've run `npm publish` from my local repo but it has not shown an update on NPM site.

closed 2020-01-23 by guymcswain