guymcswain/pigpio-client #20: lower timeout - for connection monitor

Toeedev, 2020-02-20:
as previously mentioned. Is it possible to have lower timeout so connection is monitored as frequent as possible? Currently if it goes below 2, disconnect event and connect event are triggered for no reason every couple of seconds.

Also is it possible to have timeout on, for example led.write() ? Currently, if it looses connection somewhere in between, and tries to do write(), it waits forever until connection is back. So couple of writes can pileup and take processor, as they will be stacking and waiting for connection to be back. 

(I could check if connection is lost, and have variable to check before trying to do led.write(), however connection is not checked frequently  enough)

guymcswain, 2020-02-20:
I have some questions about your use case.  Let me know if this what you are aiming for.

1.  You want to run a remote procedure, in this case represented by writing to an LED.
2.  If your remote procedure is not completed within a short period time, you want to detect this so that you can run the procedure locally (or somewhere else).
3.  The remote procedure is considered to have failed whether it was caused by:
  a) a slow network.
  b) a failed service (pigpiod).
4.  You want to retry (forever) the 'failed' remote procedure and then switch back to it.

If all you needed is 3a, I would say this is possible and perhaps a useful feature.  But I worry that you have a broader set of requirements that may not be achievable.

Toeedev, 2020-02-22:
That's exactly what I need. Main feature I need, is just to know if I lost connection to pigpiod, either through disconnected event, or during write or read action. But more frequent / quicker than 30+ seconds.