guymcswain/pigpio-client #21: AssertionError [ERR_ASSERTION]: Argument 'gpio' is not a user GPIO.

Toeedev, 2020-02-22:
As I am going along with my project. 
Why I am not allowed to use some of the GPIOs ? I do understant power GPIOs and GND GPIOs, but 
e.g. 19  or 26 ? Is it because of that mapping might be wrong?

guymcswain, 2020-02-22:
The pigpio-client follows the pigpio library convention of using the BCM GPIO numbers.  These do **not** correspond to the physical pin numbers.

Toeedev, 2020-02-22:
I do understand that, but 19 and 26 are BCM GPIO, and I use them with no problems, but your mapping is not allowing those.

guymcswain, 2020-02-22:
Can you examine and report the info object returned on the 'connected' event.  This will contain the hardware revision which is used to determine what user gpio is available.  There may be a problem in decoding the type of hardware that I have not run into.  Also, what model raspberry Pi are you using?

Toeedev, 2020-02-23:
`
{
  host: '192.168.0.21',
  port: 8888,
  pipelining: false,
  commandSocket: true,
  notificationSocket: true,
  pigpioVersion: 0,
  hwVersion: 71,
  hardware_type: 3,
  userGpioMask: 268435452,
  timeout: 1,
  version: '1.5.0'
}
`

I am using Raspberry 3B , not plus.

guymcswain, 2020-02-23:
userGpioMask: 268435452.  Hex 0FFFFFFC, means GPIO27-GPIO2 are available according to value returned by pigpio library.  This looks good.  Need to further investigate why pigpio-client thinks 19, 26 are not available.

pigpioVersion: 0.  This should be 74 for the latest version.  Do you get the same value when you run `pigs pigpv`?

timeout: 1.  Timeout should either be 0 (off) or greater than 1 to avoid race between internal timer and pigpio keep-alive packet which is sent each minute interval.  This should have nothing to do with the assertion error, just for your information.

uneasymusic, 2020-02-23:
I've changed your mask manually in info, for 0xFFFFF.. To allow me use my gpio ;)
Unless info that's displayed its something from gpiod, I will have a look on timeout

guymcswain, 2020-02-23:
I am unable to reproduce the assertion error on my Rpi3.  I read the pins without an error.

closed 2020-02-23 by guymcswain