guymcswain/pigpio-client #22: Button notify - big delay

Toeedev, 2020-02-22:
I also noticed, that `button.notify` has massive delay. I had to wait like 20-30 seconds to get event change.  I ended up checking pin manually, but maybe I am doing something wrong ? 
( I used your exact example )
Is it how it works, or I am doing it wrong. Are there any other listeners for edge?

guymcswain, 2020-02-22:
The example should work.  Something else must be going on - perhaps the network?

To see all socket transactions, including notifications, you can monitor them on a console by setting the node process environment variable `PIGPIO=1`.

Toeedev, 2020-02-23:
I am not getting any response in console from notify, only after 30 seconds maybe 
`notification received: chunk size = 12`

can't see any problems with network, I am on local network, If i check pin manually it all works fine, every second.

guymcswain, 2020-02-23:
This indicates that pigpio-client is not receiving any packets.  The problem lies somewhere else.  Perhaps in your setup.

uneasymusic, 2020-02-23:
Any, idea how I could debug that? I don't have anything additional, just pi with node on WiFi local network.

guymcswain, 2020-02-23:
A volt meter to know that you are getting the proper levels when you think the pin should report 1 or 0.

Toeedev, 2020-02-23:
when I read it manually with button.read() , every second , get's values right.

guymcswain, 2020-02-23:
Please provide a test case that shows the failure.

Toeedev, 2020-02-23:
I do get reaction from `notify`, but sometimes it takes 30 seconds and more, or it doesn't notify at all 
```javascript
const connectOptions = { host: config.slave.host, timeout: 0 } // timeout!
logger.info('Connecting to slave...')
const pigpio = client.pigpio(connectOptions) // start connecting
slave.PIR = pigpio.gpio(config.slave.pir)
slave.PIR.pullUpDown(1)
slave.PIR.modeSet('input')

slave.PIR.notify((level, tick) => {
  console.log(`Button changed to ${level} at ${tick} usec`)
})
```

guymcswain, 2020-02-23:
You need to first connect the PIR input to something you know is providing transitions between valid pin levels, in other words, pulses.

You can create a pulses by writing to the same pin for which you want to receive notifications.  

**Disconnect any external circuitry to the input pin first!**

For example, create a mock "PIR pulse":
```javascript
await slave.PIR.write(1);
await sleep(1000);  // wait one second
await slave.PIR.write(0);
```
Run this within an interval timer to get a repeating pulse train.

uneasymusic, 2020-02-23:
I am doing it by hardware, switching PIR pin between 3v3 and GND, is that not the same?

guymcswain, 2020-02-23:
Ok, so demonstrate there is a problem with the library.

Toeedev, 2020-02-23:
I don't know if it's library, but I don't do anything complicated yet. 
so I added that loop 
```javascript
slave.PIR.notify((level, tick) => {
  console.log(`Button changed to ${level} at ${tick} usec`)
})
while (true) {
    slave.PIR.read().then((read) => {
      console.log(read)
    })
    await wait(2000)
  }
```
and my console output is:
```terminal
0
0
1
1
1
1
1
1
Button changed to 1 at 193389531 usec
1
1
0
0
```
I am changing state by hardware ( jump wire from pin to either GND or 3V3 on PI ).

guymcswain, 2020-02-23:
```javascript
slave.PIR.read().then((err, read) => {
    if (err) console.log(err)  
    else console.log(read)
    })
```

guymcswain, 2020-02-23:
Sorry, you're using the promise.  So use try/catch.

guymcswain, 2020-02-23:
Run your script again using:  `PIGPIO=1 node yourscriptname.js`
Maybe slow it down to not generate so much output.

Toeedev, 2020-02-23:
ok I found the issue. I think I was setting up notify at wrong time. When I've done it on event connect. It works fine. 
I will have to make hardware pull down resistor, as it triggers loads of event on single state change. 
Well I guess it was my fault after all ;)

guymcswain, 2020-02-23:
Well, glad you got it sorted out.

guymcswain, 2020-02-23:
> ok I found the issue. I think I was setting up notify at wrong time. When I've done it on event connect. It works fine.

Makes sense.  The notify would be associated with either no socket or a socket that got closed somehow.  So if you do get a socket disconnect, you will need to re-init your notifications!

uneasymusic, 2020-02-23:
I got to that, when was reconnecting after disconnection. I have to reinit all gpios to make it work after reconnection.
This library has a big potential to create smart houses/devices. With pm2 it can be pretty reliable thing.

guymcswain, 2020-02-23:
pm2 - I'll have to lean more about it.  Thanks for the tip.

guymcswain, 2020-02-27:
I'm re-opening this issue because notifications should be self re-initialized to keep consistent with gpio objects which re-connect without requiring the application to re-initialize them.

I'm also going to change the title to make it clear this issue deals with socket disconnection and later reconnection.

I have a few other items to clean up before I release a fix in 1.5.2 in a few days.