guymcswain/pigpio-client #23: can't set up pulldown resistor

Toeedev, 2020-02-23:
I am not sure if it's me, or pulldown resistor doesn't always work. 

this is how my code looks like for setting up GPIOs, 
`const slave = {
  LED1: pigpio.gpio(config.slave.led_1),
  LED2: pigpio.gpio(config.slave.led_2),
  LEDext: pigpio.gpio(config.slave.led_ext),
  PIR: pigpio.gpio(config.slave.pir)
}

async function settingUp() {
  await slave.LED1.modeSet('output')
  await slave.LED2.modeSet('output')
  await slave.LEDext.modeSet('output')
  await slave.PIR.pullUpDown(1)
  await slave.PIR.modeSet('input')
  return
}

settingUp().then(() => {
  console.log('Slave GPIO Setup done...')
  return
})
` 
First time I got 0, but then after hours it went to 1 and stayed. 
I tried changing order of setting up, then it worked again and today stopped. 
Not pulling down. I am guessing it's something to do with me not understanding how it should work

guymcswain, 2020-02-23:
I don't see a problem with the setting up the pulldown resistor.  Keep in mind that the internal resistor is a weak resistor (probably 20K - 50K).

You should have try/catch in your async function in case of errors and in general provide an error handler for the client.

uneasymusic, 2020-02-23:
In the final system there will be harfware resistor anyway, but just for sake of testing.  I will have catch eventually ;)

guymcswain, 2020-02-23:
How is the PIR input terminated?

uneasymusic, 2020-02-23:
I am not sure what do you mean by  terminated?

guymcswain, 2020-02-23:
What is it connected to?  Physical or equivalent circuit.

uneasymusic, 2020-02-23:
For now I am only shorting it to 3v3, so there is no circuit, but you're right I know pull down up internal resistors are tricky, so that's not issue with pigpio

guymcswain, 2020-02-23:
Also, I assume somewhere else in your code you have set up a notify on the PIR gpio.

uneasymusic, 2020-02-23:
Yes and no, I was testing it with and without, and because notify wasn't working for me at the end I am not using it

closed 2020-02-23 by guymcswain