guymcswain/pigpio-client #26: Latest Installation Unit File In Git Is Wrong

dachshund-digital, 2020-06-26:
Latest installation unit file In git Is wrong, it points to the wrong location for executable.  Example unit file on information page is correct.  Please update git source for this file.

guymcswain, 2020-06-26:
Can you be more specific, I'm not following.

dachshund-digital, 2020-06-26:
Oh, sure.  If you do the typical install as per your write up pulling from git, the unit file that results points to /sbin/pigpiod.  But the actual location of the daemon is at /usr/local/pigpiod.  The example unit file you have in the your write is correct but the one deploy is incorrect.

guymcswain, 2020-06-29:
> Disregard... I posted issue to the wrong solution... the issue is with the pigpiod solution, not this client

closed 2020-06-29 by guymcswain