guymcswain/pigpio-client #28: SPI functionality

minusplusminus, 2021-03-08:
Hi,

I'm trying to connect the MPC3008, but I need SPI for this. Is there an example or something else? Or documentation so that I can add this to the lib?

Kind regards

guymcswain, 2021-03-08:
I'm currently using this wrapper for the bit-bang SPI APIs.  You can use something similar for the SPI peripheral.
```javascript
/* Wrapper for Bit Bang SPI APIs. */
exports.wrap = function (pigpio)
{
  if (pigpio.bbSPIOpen) return; // already exists

  pigpio.bbSPIOpen = function(cs, miso, mosi, sclk, baud, flags, callback)
  {
    const BSPIO = 112;
    // callback is optional
    let cb = (callback && typeof callback === 'function') ? callback : null;

    const buf = Buffer.allocUnsafe(20);
    buf.writeUint32LE(miso, 0);
    buf.writeUint32LE(mosi, 4);
    buf.writeUint32LE(sclk, 8);
    buf.writeUint32LE(baud, 12);
    buf.writeUint32LE(flags, 16);

    return pigpio.request(BSPIO, cs, 0, 20, cb, buf);
  }

  pigpio.bbSPIClose = function(cs, callback)
  {
    const BSPIC = 111;
    // callback is optional
    let cb = (callback && typeof callback === 'function') ? callback : null;
    return pigpio.request(BSPIC, cs, 0, 0, cb);
  }

  pigpio.bbSPIXfer = function(cs, data, callback)
  {
    const BSPIX = 113;
    // callback is optional
    let cb = (callback && typeof callback === 'function') ? callback : null;

    if (data) {
      let buffer = Buffer.from(data);
      return pigpio.request(BSPIX, cs, 0, data.length, cb, buffer);
    }
    else return pigpio.request(BSPIX, cs, 0, 0, cb, Buffer.alloc(0));
  }
}

```

guymcswain, 2021-03-09:
I don't see anything obvious in the nodeJS code.  If you don't have an oscilloscope/logic analyzer you can try [piscope](http://abyz.me.uk/rpi/pigpio/piscope.html).

guymcswain, 2021-03-10:
Try this:  In your callback, instead of `(e, s) => {...}` use `(e, c, s) => {...}` where c is the size (in bytes) of s.

domb84, 2022-09-24:
I'm trying to replicate my old bit of Python code using node.js and pigpio. Is this possible using this wapper? I've managed to use your wrapper to use a rotary encoder. I can't understand how to initiallise an MCP2008 though. This is what I need to replicate. Any help appreciated!

`    def buttons(self):
        # mcp3008 button reader setup
        # create software spi
        spi = bitbangio.SPI(board.D11, MISO=board.D9, MOSI=board.D10)
        # create the cs (chip select)
        cs = digitalio.DigitalInOut(board.D22)
        # create the mcp object
        mcp = MCP.MCP3008(spi, cs)
        # create analog input channels on pins 0 and 7 of the mcp3008
        chan1 = AnalogIn(mcp, MCP.P7)
        chan2 = AnalogIn(mcp, MCP.P0)

        logger.debug('MCP3008 thread start successfully, listening for buttons')

        while True:
            # read button states
            if 0 <= chan1.value <= 1000:
                self.controller.send('controls', control='timer')
            elif 5000 <= chan1.value <= 9000:
                self.controller.send('controls', control='time adj')
            elif 12000 <= chan1.value <= 15000:
                self.controller.send('controls', control='daily')
            elif 0 <= chan2.value <= 1000:
                self.controller.send('controls', control='power')
            elif 4000 <= chan2.value <= 9000:
                self.controller.send('controls', control='band')
            elif 11000 <= chan2.value <= 16000:
                self.controller.send('controls', control='function')
            elif 26000 <= chan2.value <= 29000:
                self.controller.send('controls', control='enter')
            elif 19000 <= chan2.value <= 22000:
                self.controller.send('controls', control='info')
            elif 39000 <= chan2.value <= 41000:
                self.controller.send('controls', control='auto tuning')
            elif 32000 <= chan2.value <= 34000:
                self.controller.send('controls', control='memory')
            elif 44000 <= chan2.value <= 48000:
                self.controller.send('controls', control='dimmer')
            elif chan1.value <= 64000:
                logger.debug('Uncaught press on Channel 1 %s' % chan1.value)
            elif chan2.value <= 64000:
                logger.debug('Uncaught press on  Channel 2 %s' % chan2.value)
            sleep(0.2)
`

domb84, 2022-09-25:
> No luck yet. Maybe you see something I don't see

Did you ever resolve this? I'm trying to use the wrapper function as well but none of the examples are working for me.

guymcswain, 2022-09-26:
I won’t be able to look into this in much depth for another week.  It seems your Python code is dependent on a library that you will need to implement in NodeJs to control the MCP3008.  

You may want to look at the examples, written in C, on the pigpio site.  Pigpio-client should have an equivalent api matching the c functions.

mmellema-hva, 2022-09-26:
Yes solved it.  This can be closed

minusplusminus, 2022-09-26:
Yes, above solved it. This can be closed

closed 2022-09-26 by minusplusminus

domb84, 2022-09-26:
> Yes, above solved it. This can be closed

Would you be able to share your solution?