guymcswain/pigpio-client #32: Notify on only one edge

jmbldwn, 2021-04-27:
I might just be missing it, but I can't figure out how to set up a notify for only falling or rising edges.

Is this implemented?

guymcswain, 2021-04-27:
The notifier is implemented for either edge.  Use the level argument to filter for the edge you want.  It was a decision that I made, quite a while ago, to keep things simple.

jmbldwn, 2021-04-27:
Thanks.  While it does mean twice as much traffic over the socket, I don't think it's a big deal.

jmbldwn, 2021-04-27:
Does the pigpiod socket protocol support configuring alerts for one edge?  It's hard to tell from the docs.  If it doesn't it would require changes to the daemon as well, which would be a lot of work.

jmbldwn, 2021-04-27:
Looks like the `pigs` protocol can only turn notifications on and off, no filtering on edges, so without a feature update to the daemon this isn't possible.

guymcswain, 2021-04-27:
You may be right.  The python client filters but it made do so only after having received both edges (traffic).

jmbldwn, 2021-04-27:
So this is an optimization.  The pigpio repo has a long list of issues, but might be worth adding this as a feature request.

jmbldwn, 2021-04-27:
OK, I logged an issue on pigpio for this.  No point in worrying about it here on the client until we get daemon support.

closed 2021-04-27 by jmbldwn

guymcswain, 2021-04-27:
Ok, well that will come back to me since I'm currently the maintainer of pigpio.

jmbldwn, 2021-04-27:
:-)  You're a star.  Let me know if/how I can help.