guymcswain/pigpio-client #38: writing and running pigpiod scripts?

btsimonh, 2023-01-21:
Is there a way to create and run scripts in pigpiod?

My use case is to change multiple GPOs at as close to the same time as possible.

e.g. for motor control, the following script would be useful:

let scriptID = pigpio.storeScript('w p2 0 w p4 0 pwm p0 p1 w p2 p3 w p4 p5');

pigpio.runScript(scriptID, [17, 128, 18, 0, 27, 1]);

so setting GPOs 18 & 27 to 0, then setting the pwm duty on 17 to 128, then setting 18->0 & 27->1

these actions would all occur very quickly with no network latency getting in the way....

btsimonh, 2023-01-21:
I gave a PR which implements all the script functions provided by pigpiod :).

closed 2023-01-21 by btsimonh

guymcswain, 2023-01-21:
I don’t recommend using scripts as they are not thread safe.  With pigpio, the canonical way for precise control of gpio is to use the DMA.  Look at the waveform methods.  They are driven by dma.

btsimonh, 2023-01-21:
I'm not sure I understand...

What I would like to achieve is to set values for multiple GPOs effectively simultaneously, like a salvo.  I'm not really concerned *when* they are set; what is important is that they are set together, not one after the other (within limits, obviously.  I don't need uS precision).

I do note that once I've run a script which controls certain pins, the pigpio-client has lost control of those pins.
i.e. if I have a gpio open on pin 17, and then run a script which sets pin 17, then the gpio instance in pigpio-client no longer works.
So pigpiod must be trashing the handles of the open gpios if there is a conflict, I guess...  not important from my perspective, as I won't be using the gpio instances again after initial setup (output, pwm freq, etc.).

Love the library...  very easy to use & the code is pretty understandable :).   When looking for a pigpio library, I also found https://github.com/satoren/node-pigpio - it seems comprehensive, but taking a look at how it's written, I would never have been able to understand it ....

guymcswain, 2023-01-21:
I’m traveling so getting internet connections.

Have you looked at the read/write bank apis?  You can update multiple gpio
in one command  I may not have implemented the client side api.  If so,
that would be a useful contribution you can add to the library.