guymcswain/pigpio-client #40: i2c full features

btsimonh, 2023-01-22:
I'd like to use more comprehensive i2c support, e.g. commands to read and write registers.
Use case is PCA9685 for control of servos.

Was there some reason why this is not a good addition?

(I've started coding it, but don't have much low level i2c experience, so it would be good to know if there are glaring gotchas..)
br,
Simon

guymcswain, 2023-01-22:
https://gist.github.com/guymcswain/c0f34c4cd6a38bfc2fd3108a1da3ae8d

Have a look at these wrappers for i2c.  I did then a while back and they’re
not fully tested.  I would accept a pull request if that is similar to this
style.

Thanks.

btsimonh, 2023-01-23:
Yes, the implementation follows that pattern - those functions are already in your release version, so were easy to duplicate :).
The other implementation I saw was in one of the outstanding PRs, where the OP added a i2c 'class' similar to the gpio class.  It's neater like that, but can easily be based on the first style anyway, so I'll focus on that.

For mine, I've successfully used 
```
i2cWriteByteData
i2cReadByteData
i2cWriteWordData
i2cReadWordData
```
but
`i2cWriteI2cBlockData`
leads to a later error response from pigpiod, so i've not got it quite right.

Do you know if there is a way to log communications between `pigs` and pigpiod - if so I could check the exact protocol used, and see what I'm doing wrong....

I'll PR at least the functions that work once I've used them for a while successfully.
I might also try exposing the request object so that such features *could* be added after the pigpio-client instance has been created.... which may help in development instead of modifying the library itself.

br,
Simon